Jacajack/liblightmodbus #6: Incorrectly adding modules with genconf

Wihan, 2018-02-21:
Good Day

Thanks for the great project.

I noticed that genconf.sh incorrectly adds the 02 modules when the intention is to only add 22.

To reproduce:

`./genconf.sh --endian little --mres 32 --mreq 32 --mdat 32 -m "03 04 22" -s ""`

> [info] cleaned build environment
[info] updating version number
[info] using little-endian build configuration
[info] master request buffer size is 32 bytes
[info] master response buffer size is 32 bytes
[info] master parsed data buffer size is 32 bytes
[warning] skipping slave base module
**[info] master function 2 is going to be included**
[info] master function 3 is going to be included
[info] master function 4 is going to be included
[info] master function 22 is going to be included
[info] master base is going to be included
[info] frame examination module is going to be included

Wihan, 2018-02-21:
I think the problem lies with the incorrect wildcard usage in line 106 of the genconf:

`if [[ $2 == *"$i"* ]]; then`

Jacajack, 2018-02-21:
Hi,

I'm clearly not a shell scripting master as you can see, but I'll see what I can do about that error tomorrow, when I have some spare time for googling how to properly parse strings with bash. If you already managed to replace it with something else by any chance, feel free to open a new pull request and I'll accept it.

Thank you for reporting this issue!

Wihan, 2018-02-22:
Your scripting skills are hardly anything to sell yourself short on.

The solution that I came up with was just splitting the passing in functions
`-m "03 04 22"` on a word per word basis, the for in construct in bash automatically splits on whitespace

like:
```
# for each function in the passed string, ie 3, 4, 22 in -m "3 4 22"
for argv in $2; do
    # for each available module
    for i in ${!modules[@]}; do
        # if the function passed in matches an available module exactly
        if [[ $argv == "$i" ]]; then
            printf "#define LIGHTMODBUS_F%02d%01s\n" $i $suffix >> $LIBCONF

            log "[info] $1 function $i is going to be included"

            if [[ -n ${modules[$i]} ]]; then
                echo "$1-${modules[$i]}" >> $MODCONF
                addbase=1
            fi

            break
        fi
    done
done
```

Now the problem with my solution is that now we are matching on **"3 4 22"** instead of the way you had it, "03 04 22" didn't think you'd want to change that.

Jacajack, 2018-02-23:
Yep, that seems like a much better idea than what I did. 
Thanks for contributing :+1:

closed 2018-02-23 by Jacajack