Jacajack/liblightmodbus #9: Handle invalid length of frame on slave mode

westmere, 2018-06-15:
I did some test with liblightmodbus in slave mode using serial port:
1. check slave address as a start of frame
2. with a start of frame call modbusParseRequest() on every incoming character until get a valid CRC or an exception.
In some case I got correct CRC but it is not passed length check as below:

- Valid frame: 0xa 0x10 0x0 0xd 0x0 0x1 0x2 0x0 0xac 0xd4 0x0

- Invalid frame: 0xa 0x10 0x0 0xd 0x0 0x1 0x2 0x0 0xac 0xd4
In both case it passed CRC check and return exception.
Could you suggest me a solution to handle this. Thanks you!

Jacajack, 2018-06-16:
Hi,

Thanks for reporting this issue, but I think the real problem here is the way you handle incoming Modbus frames.

Modbus relies on timings between received characters to determine where frame starts and where it ends. Received data is buffered and once there happens to be a pause longer than 3.5 characters the whole buffer should be passed to `modbusParseRequest()`.

Also, in your case the second frame is actually correct. See the picture below:

 [image]

I hope it's clear enough. I'm not home currently, but I can send you some sample Modbus receiver code tomorrow.

westmere, 2018-06-18:
Hi,
Thank you for your answer! I have already known the modbus timing, but it is still hard to implement in RTOS as the freemodbus library. Could you make the funtion modbusParseRequest() return more exception code: MODBUS_ERROR_EXCEPTION_LENGTH instead of MODBUS_ERROR_EXCEPTION. I think it will be easier to port the lib to RTOS platform.

Jacajack, 2018-06-18:
I see what you mean. In fact, I have myself already considered adding some more error macros to the library to make debugging easier. However, I don't think it's that easy.

`MODBUS_ERROR_EXCEPTION` just means that exception frame has been built, because something has gone wrong during parsing phase - the incoming frame is incorrect. It also implies that CRC happens to be valid.

Now, there's no way of knowing whether it's an actual exception or just one caused by attempt to parse  malformed frame - in both cases it's just something going wrong. If you want to get more information about the cause, you might try looking up the exception code, but I don't think it will bring you the information you are looking for, because Modbus exception set is pretty narrow as well.

To get around this problem, an extended set of build/parse errors returned in the `ModbusMaster`/`ModbusSlave` structs could be added. Each time a function would call `modbusBuildException` or return a parsing error a new value would be written, informing user what exactly has failed.

What do you think about that?

westmere, 2018-06-19:
> To get around this problem, an extended set of build/parse errors returned in the ModbusMaster/ModbusSlave structs could be added. Each time a function would call modbusBuildException or return a parsing error a new value would be written, informing user what exactly has failed.
 I think that could be a good solution, but in my opinion, i still prefer the error code more. May be the modbusParseRequestxx() function return the error when the frame length check has failed and it will not buid the exception frame. Or another way is callback mode like the freemodbus lib, but may be it will change the whole library structure.
 BTW, i think the modbus ASCII has no problem with this issue, do have a plan for the modbus ASCII support?

Jacajack, 2018-06-19:
> May be the modbusParseRequestxx() function return the error when the frame length check has failed and it will not buid the exception frame.

When frame length check fails, exception frame build process cannot be just skipped unless user explicitly commands to do so. Incomplete frame can be received due to transmission error and master must be notified about the error - the library can't just assume that you try to parse incomplete frame just because you are still receiving it.

> Or another way is implement the callback mode like the freemodbus lib, but it may change the whole library structure.

Actually, I think that the exception callback might be the best and the easiest to implement solution. I was planning to add callback  system to support user defined function codes anyway, so I believe it's good place to start.

The `modbusBuildException` would then take an additional argument describing the error with more detail. The value would then be passed to the callback function, to which pointer would be stored in `ModbusSlave` struct. That would allow easy filtering out all the length related errors.
Any further ideas?

> BTW, i think the modbus ASCII has no problem with this issue, do have a plan for the modbus ASCII support in the future?

Well, I've never given it much thought, since I only use Modbus RTU. It would require fundamental changes in the library structure, so I'm not really eager to do that right now, but if you are interested in such improvement, feel free to fork the library and experiment with the code. If you manage to get it working, I will be pleased to include the changes in the official library release.

GOST-UA, 2018-06-21:
Hi,
I really like your library, but why you use defines for exception codes and other types? IMHO, much better  will be use enumerations.
I think that the callback system for exceptions might be too complicated  - user would have to implement whole function just for error handling. I think it's just overhead, callbacks useful when we need to react differently on different data or events, but I don't see situations when I should do different actions on exception.

Jacajack, 2018-06-21:
Hi, @GOST-UA 

>I think that the callback system for exceptions might be too complicated - user would have to implement whole function just for error handling.

I didn't mean to replace the whole exception system with callback functions. I just wanted to add one called when exception frame is built - that might be useful in cases like @westmere's one. However, you are right - callbacks may be a bit overkill.

>I really like your library, but why you use defines for exception codes and other types? IMHO, much better will be use enumerations.

Well, I can't really tell you why. At some point I just decided to use defines and it just stuck that way. Enumerations are indeed a better idea and I might implement them in the future v1.4 release along with other changes that have to be done.

So... I guess that enhanced and more detailed `enum` based exception system is the compromise we are looking for. It would consist of error code returned by called function (as it is now) and then more detailed error description code in the `ModbusMaster`/`ModbusSlave` structure. What do you think about that? Do you have any further ideas and recommendations?

GOST-UA, 2018-06-22:
I think it might be good.
Also just today I got some troubles with modbusBuildRequest03 (-04): they declared as defines and there are no way to use function pointers on it. So I must create my own wrappers as a workaround. I think `static inline` functions in header file should be better solution. 
And one more thing: I think callback system would be good solution for creating something like lookup table for Modbus Slave. This will allow user to configure Slave device just specifying start register, length and callback function that will handle request for particular address range.

P.S. Sorry if my English look bad ))

GOST-UA, 2018-06-22:
Oh, I also remember another one idea - Modbus TCP. In one of my future project I'll need it, so I could try implement Modbus TCP like part of this library. What do you think about this?

Jacajack, 2018-06-22:
> Also just today I got some troubles with modbusBuildRequest03 (-04): they declared as defines and there are no way to use function pointers on it. So I must create my own wrappers as a workaround. I think static inline functions in header file should be better solution.

Sorry for that. I never thought that one may need to have pointer to these functions, so I just decided to do it the easiest way possible. Apparently this needs fixing as well.

> I think callback system would be good solution for creating something like lookup table for Modbus Slave.

I thought about something similar, with possibility to address discontinuous register space, but that would require a lot more work. This sounds like good place to start.

I never used Modbus TCP and I don't know a lot about it, however, if you want to try to implement it - sure, go ahead. I believe that the main library code can (and should) be reused, so I'd suggest implementing it as an optional [add-on](https://github.com/Jacajack/liblightmodbus/tree/master/src/addons).

I think I can start working on the proposed changes on the devel branch after the weekend. New ideas and pull requests are obviously welcome :smile:

P.S. Your English is absolutely fine :)

Jacajack, 2018-06-30:
@GOST-UA I added the discussed functionalities to the library. The stable changes (register/coil callbacks, user defined functions, disabling master data buffer) are merged into master branch, so you can take a look if you want.

I'm still thinking on the improved error handling, but the changes are to come soon too. Please let me know if you have any suggestions.

GOST-UA, 2018-08-14:
Hi @Jacajack! Currently I can't look closely on newest changes and make feature that I committed.
 But one of my colleagues use your library and has short feedback:

1. We use library in C++ project, so it will be nice if you add `extern "C"` to library public headers.
2. We also use RTOS system and callback system is a little uncomfortable because there is no way to pass task context throw callback. I think it can be solved by adding extra parameter `void *ctx` to callback signature with method to set this parameter.

And one note from me: do You hear something  about [Keep a Changelog](https://keepachangelog.com/en/1.0.0/) and [Semantic Versioning](https://semver.org/)? I think following by this  two documents can make library a bit more friendly and clearly.

Jacajack, 2018-08-14:
Hey @GOST-UA! I thought this thread was dead, I'm glad to see you back.

These 2 changes seem very reasonable, I'll add them when I have time.

Regarding changelog - [the changelog](https://github.com/Jacajack/liblightmodbus/blob/packaging/debian/changelog) is available on the packaging branch, because it's needed for Launchpad. Also, I believe starting semantic versioning (I've only heard this name and never used it) is better late than never, so this release feels like a good place to start. In this case, I think it should be renamed to `v2.0`.

GOST-UA, 2018-08-22:
I think we should leave this thread dead 😀. We talk about different things that doesn't relate to the issue long time. Should we start new issue each time when talking about new problems/features or there is another way to discuss some moments before open issue?

Jacajack, 2018-08-22:
Yeah, we've gone off topic, that's for sure. :stuck_out_tongue_winking_eye:

Yes, I think that's the best way to go. Feel free to open new issues and pull requests whenever you have any concerns or ideas. You can also email me, if you want.

Since I believe that most recent changes made to the library address this issue and make error distinguishing easier, I think this issue can be closed, at least for now.

closed 2018-08-22 by Jacajack