Jacajack/liblightmodbus #11: How to use liblightmodbus with STM32?

Linjieqiang, 2019-10-31:


Jacajack, 2019-10-31:
1. Clone the library from Github - `git clone https://github.com/Jacajack/liblightmodbus.git`
2. Create build directory and enter it - `mkdir build && cd build`
3. Run CMake with these flags:
    - `CC` must be path to your ARM compiler (e.g. `arm-none-eabi-gcc`)
    - `LD` must be path to your ARM linker (e.g. `arm-none-eabi-ld`)
    - `CMAKE_C_FLAGS` should contain compilation flags necessary for your application
4. The command should look like this: `cmake .. -DCC=arm-none-eabi-gcc -DLD=arm-none-eabi-ld -DCMAKE_C_FLAGS="-Wall -mthumb and so on..."`
5. Afer CMake completes its job successfully, you can build the library by running `make`
6. Add the library to your project
    - Add `lightmodbus/include` to your include directories
    - Add `lightmodbus/build` to linker search paths or add `lightmodbus/build/liblightmodbus.a` as a source file

Jacajack, 2019-10-31:
Closing, but feel free to comment

closed 2019-10-31 by Jacajack

Linjieqiang, 2019-10-31:
@Jacajack Hello, But should I implement some api functions?

Jacajack, 2019-10-31:
Hello,

Liblightmodbus is only capable of parsing and creating Modbus protocol data frames.
How the frames are transmitted is a platform-dependent issue, so you will have to create your own functions for communication, if that's what you mean.

That's not really hard though. The trickiest part is correctly splitting incoming frames. [Here's](https://gist.github.com/Jacajack/17faae3283ee943ab081f0b22b4b52f3) my old implementation for STM32. It's not great, but it can be useful for you.