Jacajack/liblightmodbus #14: Getting segmentation fault when using statically alllocated buffers

ASL07, 2020-02-11:
Hi

Thanks for the great library. 

When dynamic memory allocation is disabled, I get a segmentation fault in the modbusMasterInit() function. I cannot see anything wrong with it after trying to debug for several hours (I might be doing something wrong somewhere). I can reproduce it in both Os X and debian.
This issue disappears once I enable dynamic memory allocation

I have created a project in order to reproduce and isolate the problem as much as possible in this repo https://github.com/ASL07/lighmodbus. 
If you run the build script, it will trigger the execution of the test main(). It will throw a segmentation error after the program ends.

If  `add_definitions(
    -DLIGHTMODBUS_STATIC_MEM_SLAVE_REQUEST=256
    -DLIGHTMODBUS_STATIC_MEM_SLAVE_RESPONSE=256
    -DLIGHTMODBUS_STATIC_MEM_MASTER_REQUEST=256  
    -DLIGHTMODBUS_STATIC_MEM_MASTER_RESPONSE=256   
    -DLIGHTMODBUS_STATIC_MEM_MASTER_DATA=256    
)` is comented out in CMakeLists, then the issue disappears, however, once these macros are defined, then we get the error again.

I have tried to modify the size of the buffers and it makes no change either.

Could you please have a look into it?

Regards

Jacajack, 2020-02-11:
Hi,

I can reproduce the error in the repo you provided, but I don't think it is an error in the library itself.

In your CMakeLists you have this:
```cmake
add_definitions(
    -DLIGHTMODBUS_STATIC_MEM_SLAVE_REQUEST=256
    -DLIGHTMODBUS_STATIC_MEM_SLAVE_RESPONSE=256
    -DLIGHTMODBUS_STATIC_MEM_MASTER_REQUEST=256  
    -DLIGHTMODBUS_STATIC_MEM_MASTER_RESPONSE=256   
    -DLIGHTMODBUS_STATIC_MEM_MASTER_DATA=256    
)
```
Which indeed makes the compiler define the macros during compilation of the library, but does not put proper `#define`s in the library configuration file `include/lightmodbus/libconf.h`:
```c
#ifndef LIGHTMODBUS_LIBCONF_H
#define LIGHTMODBUS_LIBCONF_H

#define LIGHTMODBUS_VERSION "2.0"
#define LIGHTMODBUS_GIT_VERSION "524828-dirty"
#define LIGHTMODBUS_MASTER_BASE
#define LIGHTMODBUS_F01M
#define LIGHTMODBUS_F02M
#define LIGHTMODBUS_F03M
#define LIGHTMODBUS_F04M
#define LIGHTMODBUS_F05M
#define LIGHTMODBUS_F06M
#define LIGHTMODBUS_F15M
#define LIGHTMODBUS_F16M
#define LIGHTMODBUS_LITTLE_ENDIAN

#endif
```

This causes the library to use different definitons of data structures than your code, which results in stack smashing. 

Instead, you should be defining CMake variables (such as `MODULES`), so it can generate the config file for you. I just realized that the documentation can be a bit misleading, because it gives following example:
```cmake -DSTATIC_MEM_SLAVE_REQUEST=128 ...```
which might suggest that `-DSTATIC_MEM_SLAVE_REQUEST` is a compiler flag, which it is not.

Try this instead, please:
```cmake
set(STATIC_MEM_SLAVE_REQUEST 256)
set(STATIC_MEM_SLAVE_RESPONSE 256)
set(STATIC_MEM_MASTER_REQUEST 256)
set(STATIC_MEM_MASTER_RESPONSE 256)
set(STATIC_MEM_MASTER_DATA 256)
set(MODULES MASTER_BASE;F01M;F02M;F03M;F04M;F05M;F06M;F15M;F16M)
add_subdirectory(extern/liblightmodbus)
```

Resulting `libconf.h` should look like this:
```c
#ifndef LIGHTMODBUS_LIBCONF_H
#define LIGHTMODBUS_LIBCONF_H

#define LIGHTMODBUS_VERSION "2.0"
#define LIGHTMODBUS_GIT_VERSION "524828-dirty"
#define LIGHTMODBUS_MASTER_BASE
#define LIGHTMODBUS_F01M
#define LIGHTMODBUS_F02M
#define LIGHTMODBUS_F03M
#define LIGHTMODBUS_F04M
#define LIGHTMODBUS_F05M
#define LIGHTMODBUS_F06M
#define LIGHTMODBUS_F15M
#define LIGHTMODBUS_F16M
#define LIGHTMODBUS_STATIC_MEM_SLAVE_REQUEST 256
#define LIGHTMODBUS_STATIC_MEM_SLAVE_RESPONSE 256
#define LIGHTMODBUS_STATIC_MEM_MASTER_REQUEST 256
#define LIGHTMODBUS_STATIC_MEM_MASTER_RESPONSE 256
#define LIGHTMODBUS_STATIC_MEM_MASTER_DATA 256
#define LIGHTMODBUS_LITTLE_ENDIAN

#endif
```

Let me know it helps
Regards

ASL07, 2020-02-12:
@Jacajack 
That worked perfectly. I suspected there was some problem with the build configuration but I wasn't able to find it.
Didn't realize the definitions needed to be set as Cmake variables instead of compiler flags
Thank you for the quick response

closed 2020-02-12 by ASL07

Jacajack, 2020-02-12:
I'm glad I could help. I'll improve the documentation on static memory allocation, so it doesn't mislead anyone else.