Jacajack/liblightmodbus #17: Adding build support for ESP-IDF

Maldus512, 2020-10-21:
Hello,
I've found your library very useful as a lightweight, hardware-agnostic library to handle Modbus communication, and I've integrated it into a project for an ESP32-powered device.

The official framework for ESP32 ([ESP-IDF](https://github.com/espressif/esp-idf)) is mostly CMake-based, so I believe it would be easy to integrate liblightmodbus as a component for ESP-IDF.

Basically the build system looks for `CMakeLists.txt` files [that describe attributes for the component ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html)(source files, include dirs, etc...). For now I'm including liblightmodbus as a git submodule in my project and refer to it with a different `CMakeLists.txt`:

```
components
    /liblightmodbus/CMakeLists.txt    <- custom component descriptor written by me to include liblightmodbus into an ESP-IDF project
    /liblightmodbus/liblightmodbus     <- actual Git repository
```

It would be easier for me to just clone liblightmodbus into the `components` folder and expect its `CMakeLists.txt` to work. 
Since I've seen there is already some distinctions in it based on an `AVR` variable, would you agree to adding something similar for ESP32? The ESP-IDF framework defines the `ESP_PLATFORM` variable when building, so I've found that simply adding the following lines to the very beginning of `CMakeLists.txt` make it work (at a very basic level):

```
if (ESP_PLATFORM) 
    idf_component_register(
        SRC_DIRS src src/slave src/master
        INCLUDE_DIRS include .
    )
else()
    return()
endif()
```

This cuts off all of the configuration checks and `libconf.h` generation, so that part still needs to be addressed. If you were okay with that I could work a little more on this extension and create a pull request.

Jacajack, 2020-10-22:
Hello,

I'm very happy to hear that you've found liblightmodbus useful and want to make it more accessible. Personally I've never used ESP32 or any of its frameworks, so unfortunately I don't have any experience in this area. But if you think you could indeed make liblightmodbus easier to integrate with ESP-IDF simply by adjusting `CMakeLists.txt`, you have my support. Please feel free to open a pull request, if you manage to get this working. :+1: 

Thank you!

Maldus512, 2020-10-24:
Great! I have opened a pull request (#18 ) with the basic modifications in place. If it looks good to you I'll refine the last details (mainly the descriptions for the kconfig options) and give it a test round on a real device.

Please note that I am but a novice with CMake, so my work could be sloppy. Do not hesitate to correct me if that is the case.
The remaining functionality of the library should be unchanged, I have checked the example and it still works.

closed 2020-10-27 by Jacajack

Jacajack, 2021-08-07:
Since v3.0 liblightmodbus is a header-only lilbrary. I had a bit of trouble with accommodating ESP-IDF support for that change, but I think I eventually managed to come up with a quite good solution.

I decided that it would be better to keep all the ESP-IDF files on a separate branch as not to confuse people with that `CMakeLists.txt` only meant for registering a component. Now all ESP-IDF files are on the [`esp-idf`](https://github.com/Jacajack/liblightmodbus/tree/esp-idf) branch. It can be cloned directly to your components directory using:
```
git clone --single-branch --branch esp-idf https://github.com/Jacajack/liblightmodbus.git
```
The CMake configuration takes care of downloading, configuring and building the newest version. There's no need to explicitly include liblightmodbus implementation in your source code, because the component registers a source file including it for you. You shouldn't define any config options when including `lightmodbus.h` either, because the entire configuration is read from `config.esp.h` and `sdkconfig.h` and should be edited with `idf.py menuconfig`.

I'm barely an ESP novice, but I hope I did everything right. @Maldus512, maybe you have some suggestions?

Maldus512, 2021-08-10:
First of all thanks for keeping up the support for ESP32; although I haven't followed the development discussion for liblightmodbus v3.0 but I am still using v2.0 regularly on ESP32 projects.

I had proposed to integrate the build system into the lightmodbus CMake because it was already there; since it is now a include only library I agree leaving the extra `CMakelists.txt` would just be confusing.

I would probably have created a separate git repository that includes liblightmodbus as a submodule and does exactly what the `esp-idf` branch does, but I don't see any downside to your approach. I'll certainly try the new version in the following months and will give feedback if I found anything problematic.

Jacajack, 2021-08-10:
Creating a separate repository might in fact be a better idea. Now I'm just wondering whether to use a git submodule or `FetchContent_Populate()` in CMake.

Thank you for your response. I'm glad to hear you're planning to continue using libligthmodbus. Any feedback will be very welcome.