Jacajack/liblightmodbus #23: is there any template/port file for mcu UART?

sushant-ht, 2021-08-05:
Hi,
Just wanna try this library, is there any project template for using it with UART of any MCU? just how to add the library to the project & implement UART routines to call the library functions. 

is their master example also, all have seen is slave test files.

Jacajack, 2021-08-05:
Hi,

At the moment - no, at least not officially, but thank you for the idea. It feels like something that should definitely be added in the upcoming v3.0 release. If you like, I can provide you my old implementation for STM32, though.

>is their master example also, all have seen is slave test files.

[This file](https://github.com/Jacajack/liblightmodbus/blob/master/examples/example.c) imitates communication process between  master and slave, including building requests using `modbusBuildRequest*()` functions.

Hope this helps and thank you!

Jacajack, 2021-08-09:
I added a fully-featured AVR slave example for v3.0 [here](https://github.com/Jacajack/liblightmodbus/tree/dev-v3.0/examples/avrslave). Code for parsing Modbus frames looks a bit different than in older versions, but the rest (interrupts, timeout calculation, etc.) is exactly the same.

closed 2021-08-09 by Jacajack

sushant-ht, 2021-08-12:
Thanks for quick response