Jacajack/liblightmodbus #26: For loop initial declarations are incompatible with XC16

Maldus512, 2021-09-15:
Hello,
I have started using liblightmodbus 3.0 in some of my software projects; initially I was successful, but I was immediately halted while trying to port to the new version a PIC24 project.
The default behavior of the XC16 compiler from Microchip forbids variable declarations inside a for header (in compliance with the C90 standard I believe), which liblightmodbus does in a few points ([i.e. here](https://github.com/Jacajack/liblightmodbus/blob/e7be88bc65abec4a902f4e5194d7235ebd3a19aa/include/lightmodbus/slave.impl.h#L254)). Therefore my compilation fails with the error `error: 'for' loop initial declarations are only allowed in C99 mode`.

Trying to use C99 mode with the `-std=c99` flag causes some other issues in my project which I did not yet explore extensively. Personally I like to declare indexes inside the for header as well, but the PIC compiler often leaves me with little choice. I'm here to ask if you would be willing to accept a pull request where variable declarations have been moved outside the for statement, in which case I'll see to it. Otherwise I'll keep trying to use "C99 mode" and fix the other issues that arise with that.

Jacajack, 2021-09-15:
Hello,

Oh damn... I didn't expect that. I assumed that switching to C99 wouldn't be an issue with modern compilers, but well... here we are.

This quite problematic, because even if we changed all the `for` loops, the library still wouldn't work with a strictly C89 compiler. One reason is the use of compound literals:
https://github.com/Jacajack/liblightmodbus/blob/e7be88bc65abec4a902f4e5194d7235ebd3a19aa/include/lightmodbus/base.h#L80

Then you have "mixed declarations and code" which also are theoretically C89 non-compliant:
https://github.com/Jacajack/liblightmodbus/blob/e7be88bc65abec4a902f4e5194d7235ebd3a19aa/include/lightmodbus/master_func.impl.h#L410-L417

Having said that, the `for` loops are the only thing actually preventing me from compiling liblightmodbus v3.0 with `--std=c89 -Wpedantic` on gcc. Everything else is just (a lot of) warnings. As much as I dislike the idea of mixing C89 loops with C99 code, this makes me think that such PR could be acceptable as a last resort.

I really don't want to cause you unnecessary work, but could you please start with an attempt to make your code C99 compatible? The subset of things that work in C89 and don't in C99 is rather limited (unless it's some another XC16 oddity). [The C Rationale](http://www.open-std.org/jtc1/sc22/wg14/www/C99RationaleV5.10.pdf) might come very useful here - just search the document for "quiet change in c99". Maybe your code contains some UB triggered by different compiler settings? Personally, I think that might be worth exploring.

Alternatively, since all the `for` loops are in `.impl.h` files, I think you could try to compile the main library implementation source file as C99 and the rest of your code as C89? This seems to be a good compromise that might actually work.

If you're unable to track down the problem and the second approach doesn't help you either, we could start thinking about rewriting the `for` loops. For now, I wish you luck and hope I could help somehow.

Maldus512, 2021-09-15:
> I really don't want to cause you unnecessary work, but could you please start with an attempt to make your code C99 compatible? 
No worries, I asked exactly for this reason.

> The subset of things that work in C89 and don't in C99 is rather limited (unless it's some another XC16 oddity).
I think XC16 doesn't strictly follow a single, clearly defined standard but I'll look into it.

> Alternatively, since all the for loops are in .impl.h files, I think you could try to compile the main library implementation source file as C99 and the rest of your code as C89? This seems to be a good compromise that might actually work.
I didn't consider this, I'll check it as well. As a last resort I could include liblightmodbus as a separately compiled library, I'm quite sure that would allow me to set separate compilation flags.

> If you're unable to track down the problem and the second approach doesn't help you either, we could start thinking about rewriting the for loops. For now, I wish you luck and hope I could help somehow.
I know it's not a straightforward request, that's why I opened an issue instead of jumping right into a pull request. I'll study my situation a little more. In the meantime thanks for the insight!

Maldus512, 2021-09-17:
I was successful in porting my code to C99 mode. 
The issue in my code was with parenthesized compound statements, which are not accepted by XC16 in C99 mode. My knowledge of C standards is limited but I was confused by the fact that compound statements are accepted by default (C89) but refused in C99 mode. I then found out that they are a GCC extension, so I guess it is less absurd?
Anyway, as you said it is better to adapt my code than to chase C89 compliance for liblightmodbus, and the issue is solved for me. Thanks for the assistance!

closed 2021-09-17 by Maldus512

Jacajack, 2021-09-17:
That sounds very strange indeed. If [this](https://stackoverflow.com/questions/50353399/compound-expression-in-if-statement) is what you mean, I had completely no idea you could do that with GCC. Always good to learn something new, though.

I'm glad to hear you worked it out! :+1: