Jacajack/liblightmodbus #27: Advise to implement function 21 (0x15) Write File Record

TacHuynh, 2021-11-24:
Please advise on best knowledge ideas/where to start to implement  function 21 (0x15) Write File Record
Thanks in advance.
Tac.

Jacajack, 2021-11-26:
You will need to implement a couple of functions: `modbusParseRequest21` in [slave_func.h](https://github.com/Jacajack/liblightmodbus/blob/master/include/lightmodbus/slave_func.h), `modbusBuildRequest21` and `modbusParseResponse21` + a set of macros in [master_func.h](https://github.com/Jacajack/liblightmodbus/blob/master/include/lightmodbus/master_func.h). You can use the existing functions as a reference. You will also have to add an additional field to `ModbusDataCallbackArgs` and `ModbusRegisterCallbackArgs` for passing the file ID. If you implement function 21, you'll probably want to have 20 as well, so multiply all that by two. 

As per implementation details, I can't really help you here, because I haven't implemented it myself (as you can see). I initially wanted to include 20 and 21 in v3.0, but eventually decided to postpone that. When in doubt, I would recommend sticking to the [Modbus Application Protocol](https://modbus.org/docs/Modbus_Application_Protocol_V1_1b.pdf). You will definitely have to make some rather difficult design decisions (e.g. how the file contents are passed to the callback functions). Getting all the safety checks right might be a bit tricky too.

Hope this helps! Good luck!

TacHuynh, 2021-11-27:
Thank you for your pointers.
I will try to implement these functions per your advise.
Tac.

closed 2021-12-07 by Jacajack