Jacajack/liblightmodbus #28: Example form TCP master integration

rtborg, 2022-04-30:
Hi, I'm looking for example of integrating Modbus TC master functionality, currently I'm using `espidf` but any example will do.

Thanks

Jacajack, 2022-05-06:
Hi,

A simple example for Modbus RTU master application is available [here](https://github.com/Jacajack/liblightmodbus/blob/master/examples/linuxmaster/master.c). Creating Modbus TCP data frames should be mostly a matter of replacing `RTU` with `TCP` in function calls.

If you're looking for an example implementation of a full Modbus TCP client/server application with liblightmodbus, sadly, there are none. Good thing that you pointed that out. When I have some time, I will try to create an example TCP client/server pair for Linux. The implementation for ESP32 will likely be very similar.

As for now, I'm afraid you'll have to dive into [Modbus TCP Implementation Guide](https://www.modbus.org/docs/Modbus_Messaging_Implementation_Guide_V1_0b.pdf) and the ESP socket API. If you have any success with getting Modbus TCP to work on the ESP, please feel free to open a pull request and submit some example code. I'm sure it will be useful.

Sorry for my late response. Hope this helps!

rtborg, 2022-05-10:
Thank you for the response. I managed to integrate `liblightmodbus` in a FreeRTOS projec with using BCD sockets and `lwIP`. As I needed only reading functionality, it was pretty easy.  
Hope to find time to share code soon.

closed 2022-05-10 by rtborg

Jacajack, 2022-05-10:
Glad to hear that!