Jacajack/liblightmodbus #29: Assistance with TCP usage

ammaree, 2022-07-19:
@Jacajack 
I am trying to make your library work with the ESP32 using ESP-IDF.

Do you by any chance have a TCP example that I can use as a starting point?

ammaree, 2022-07-19:
Thanks, problem resolved.

However, the PLC's we are using have large numbers of u32, i32 and f32 registers.

Any plans to add support for 32bit register sizes with these formats?

Jacajack, 2022-07-19:
Hello,

Sadly, currently there's no real example for Modbus TCP. This has already been discussed in #28. I'm really going to have to look into that. Glad to hear you already managed to solve your problem, though.

As far as I know, the implementation of the 32-bit types in Modbus is non-standard and might differ between manufacturers. Because of that, I don't think integrating it in the core library would be a good idea. However, having a couple of additional vendor-specific implementations of these features in separate files might be beneficial. I'm not planning to implement them myself, but I would gladly accept any pull requests with such 'extensions' to the library.

There's a little catch though - to handle the 32-bit types properly, support for wider data types would first have to be added to [`ModbusRegisterCallbackArgs`](https://jacajack.github.io/liblightmodbus/structModbusRegisterCallbackArgs.html). This sounds like something that should be implemented alongside with support for file read/write operations. I'm planning to do that in the near future, maybe even this summer, although there's no clear deadline yet.

ammaree, 2022-07-26:
Hi @Jacajack 

Overall a superb, simple and compact library, far easier to get started with compared to the ESP-IDF ESP-Modbus component. 

I have added functionality in the data callback that matches the index with a set of index/count entries and, if the count is >1 consolidates successive 16bit values into a single 32/64 bit value. I have also added flags that allow for specification of:
a) un/signed or signed or float format;
b) little or big endian byte order ie. reversed if required
So far this is working well with both Carlo Gavazzi energy meters and Eliwell/Schneider PLC's

However, my solution is a kludge and does not nicely integrates with the format of your library so I am looking forward to you adding proper support for 32bit, maybe even 64 bit types as well as the file/read support.

Thanks

André

Jacajack, 2022-07-27:
I'm happy to hear you like the library, thanks!

Good to hear you managed to find a way around that limitation. Maybe it's not a perfect solution, but definitely is a good one, if you managed to get it working with real hardware.

I will look into providing real support for that soon. I'd be great if I could find a way that combines both file read/write and wide data types support into a single non-breaking change. Let's keep this issue open until that gets done - I will try to keep you updated here.

Silvesterrr, 2025-03-29:
@ammaree Do you maybe have modbus tcp slave example using esp-idf? 
I've integrated esp-idf modbus rtu  master. Building the library was much easier using liblightmodbus.

@Jacajack 
My only sugestion is that it would be nice if you could directly get parsing result from ```modbusParseResponseRTU``` without needing to use calback handler. :)

Jacajack, 2025-03-29:
> My only sugestion is that it would be nice if you could directly get parsing result from modbusParseResponseRTU without needing to use calback handler. :)

@Silvesterrr Thanks, that's a good point. Do I understand correctly that you'd like to be able to call `modbusParseResponseRTU` while `dataCallback` is `NULL` in the `ModbusMaster` struct? If so, that should be relatively easy to implement.

Jacajack, 2025-03-29:
Ah, right, I see. Actually, I deliberately avoided that approach. The reason for that is that would require allocating an additional buffer for the processed values. That's not very desirable on low memory systems, especially considering that you can have up to 2000 coil values in a response. Sacrificing 2kB of memory for some user convenience didn't feel like a good decision to make. That's why I went with the more flexible and memory-light callback approach.

However, something like that can be achieved quite easily with the current system, if you like. You'll just need to write a small callback which puts all the data in some array of your choice. Something like this:
https://github.com/Jacajack/liblightmodbus/blob/00356018db0f7971a1a66ae810b6bab48ec756cb/test/tester.cpp#L123-L129

>Overall the library is great thx for that :).

Glad to hear you like it, thanks :)

ammaree, 2025-03-31:
@Silvesterrr

Have only implemented a Modbus Master, no need for slave (yet)