Jacajack/liblightmodbus #32: Can be added to vcpkg? I have difficulty compiling in vs2022

icyfox168168, 2023-03-06:
I am very interested in testing this library

Jacajack, 2023-03-06:
Hi, I am not familiar with vcpkg, but you should be able to build in VS2022. What seems to be the problem exactly?

icyfox168168, 2023-03-07:
1>cppdemo.cpp
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(288,1): error C4430: 缺少类型说明符 - 假定为 int。注意: C++ 不支持默认 int
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(288,1): error C2065: “warn_unused_result”: 未声明的标识符
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(288,25): error C2146: 语法错误: 缺少“;”(在标识符“ModbusError”的前面)
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(293,1): error C4430: 缺少类型说明符 - 假定为 int。注意: C++ 不支持默认 int
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(293,1): error C2374: “__attribute__”: 重定义；多次初始化
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(288,1): message : 参见“__attribute__”的声明
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(293,1): error C2065: “warn_unused_result”: 未声明的标识符
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(293,1): error C2146: 语法错误: 缺少“;”(在标识符“ModbusErrorInfo”的前面)
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(295,1): error C4430: 缺少类型说明符 - 假定为 int。注意: C++ 不支持默认 int
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(295,1): error C2374: “__attribute__”: 重定义；多次初始化
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(288,1): message : 参见“__attribute__”的声明
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(295,1): error C2065: “warn_unused_result”: 未声明的标识符
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(295,25): error C2146: 语法错误: 缺少“;”(在标识符“ModbusError”的前面)
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(333,1): error C4430: 缺少类型说明符 - 假定为 int。注意: C++ 不支持默认 int
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(333,1): error C2374: “__attribute__”: 重定义；多次初始化
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(288,1): message : 参见“__attribute__”的声明
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(333,1): error C2065: “warn_unused_result”: 未声明的标识符
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(333,25): error C2062: 意外的类型“unsigned char”
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(334,1): error C2143: 语法错误: 缺少“;”(在“{”的前面)
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(334,1): error C2447: “{”: 缺少函数标题(是否是老式的形式表?)
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(357,1): error C4430: 缺少类型说明符 - 假定为 int。注意: C++ 不支持默认 int
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(357,1): error C2374: “__attribute__”: 重定义；多次初始化
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(288,1): message : 参见“__attribute__”的声明
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(357,1): error C2065: “warn_unused_result”: 未声明的标识符
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(357,25): error C2062: 意外的类型“unsigned short”
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(358,1): error C2143: 语法错误: 缺少“;”(在“{”的前面)
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(358,1): error C2447: “{”: 缺少函数标题(是否是老式的形式表?)
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(365,1): error C4430: 缺少类型说明符 - 假定为 int。注意: C++ 不支持默认 int
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(365,1): error C2374: “__attribute__”: 重定义；多次初始化
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(288,1): message : 参见“__attribute__”的声明
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(365,1): error C2065: “warn_unused_result”: 未声明的标识符
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(365,25): error C2062: 意外的类型“unsigned short”
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(366,1): error C2143: 语法错误: 缺少“;”(在“{”的前面)
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(366,1): error C2447: “{”: 缺少函数标题(是否是老式的形式表?)
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(385,1): error C4430: 缺少类型说明符 - 假定为 int。注意: C++ 不支持默认 int
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(385,1): error C2374: “__attribute__”: 重定义；多次初始化
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(288,1): message : 参见“__attribute__”的声明
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(385,1): error C2065: “warn_unused_result”: 未声明的标识符
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(385,25): error C2062: 意外的类型“unsigned short”
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(386,1): error C2143: 语法错误: 缺少“;”(在“{”的前面)
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(386,1): error C2447: “{”: 缺少函数标题(是否是老式的形式表?)
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(422,1): error C4430: 缺少类型说明符 - 假定为 int。注意: C++ 不支持默认 int
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(422,1): error C2374: “__attribute__”: 重定义；多次初始化
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(288,1): message : 参见“__attribute__”的声明
[... cut ...]
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.h(138,1): error C4430: 缺少类型说明符 - 假定为 int。注意: C++ 不支持默认 int
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.h(138,1): error C2374: “__attribute__”: 重定义；多次初始化
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.h(288,1): message : 参见“__attribute__”的声明
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.h(138,1): error C2065: “warn_unused_result”: 未声明的标识符
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.h(138,1): fatal  error C1003: 错误计数超过 100；正在停止编译

icyfox168168, 2023-03-07:
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.impl.h(60,2): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.impl.h(61,9): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.impl.h(128,10): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.impl.h(133,9): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.impl.h(155,9): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.impl.h(175,10): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.impl.h(194,9): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.impl.h(232,9): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.impl.h(279,10): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.impl.h(315,10): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.impl.h(319,10): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.impl.h(334,11): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.impl.h(344,11): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.impl.h(347,9): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.impl.h(315,10): error C2397: 从“ModbusError”转换到“unsigned int”需要收缩转换
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.impl.h(344,11): error C2397: 从“ModbusError”转换到“unsigned int”需要收缩转换
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.impl.h(379,10): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.impl.h(399,11): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.impl.h(402,9): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.impl.h(379,10): error C2397: 从“ModbusError”转换到“unsigned int”需要收缩转换
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.impl.h(399,11): error C2397: 从“ModbusError”转换到“unsigned int”需要收缩转换
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave_func.impl.h(61,11): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave_func.impl.h(99,10): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave_func.impl.h(120,9): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave_func.impl.h(173,10): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave_func.impl.h(179,9): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave_func.impl.h(253,10): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave_func.impl.h(259,9): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave_func.impl.h(323,10): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave_func.impl.h(330,9): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\master.impl.h(114,9): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\master.impl.h(123,9): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\master.impl.h(133,9): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\master.impl.h(149,10): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\master.impl.h(151,9): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\master.impl.h(149,10): error C2397: 从“ModbusError”转换到“unsigned int”需要收缩转换
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\master.impl.h(161,9): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\master.impl.h(180,10): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\master.impl.h(182,9): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\master.impl.h(180,10): error C2397: 从“ModbusError”转换到“unsigned int”需要收缩转换
[... cut ...]
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\master_func.impl.h(417,10): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\master_func.impl.h(427,9): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\master_func.impl.h(445,10): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\master_func.impl.h(452,9): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法

icyfox168168, 2023-03-07:
Can you take time to update? I really need this library

Jacajack, 2023-03-07:
I think that one issue might be that MSVC doesn't understand attributes meant for `clang` and `gcc`. Please try to add this code before including any library headers:
```
#define LIGHTMODBUS_WARN_UNUSED
```

You can also add this empty definition in compilation settings.

Also, it seems that your compiler is treating all warnings as errors. I'm not sure if you want that kind of behavior.

icyfox168168, 2023-03-08:
Still not, according to your method, I use the latest version of C++20 standard vs2022

Jacajack, 2023-03-08:
Did you try disabling treating warnings as errors? What are the error messages now?

icyfox168168, 2023-03-08:
The first error window has been resolved, and the error of this window remains

1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.impl.h(60,2): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\base.impl.h(61,9): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.impl.h(128,10): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法
1>D:\Users\fix\Documents\GitHub\liblightmodbus\include\lightmodbus\slave.impl.h(133,9): error C4576: 后跟初始值设定项列表的带圆括号类型是一个非标准的显式类型转换语法

icyfox168168, 2023-03-08:
My warning level is only 1. My environment is win11+vs2022+qt5.15,8 My cpp example is compiled

Jacajack, 2023-03-08:
Okay, so it doesn't recognize compound literal syntax. It's quite strange, because that's standard C99 code (see section 6.5.2.5.11 [here](https://www.open-std.org/jtc1/sc22/wg14/www/docs/n1256.pdf)). It seems to be a Microsoft C++ compiler quirk - https://stackoverflow.com/a/3870137/4019966.

I suppose the solution is to include liblightmodbus implementation ( `LIGHTMODBUS_IMPL`) in a C file instead (`.c` and not `.cpp`). Of course, you should still be able to include `liblightmodbus.h` and call functions from C++ files. Let me know if that works for you.

icyfox168168, 2023-03-09:
I switched to LLvm compilation in vs2022 and it was fine