Jacajack/liblightmodbus #34: modbusMasterInit() and LPC1768

PoojaEmbedded, 2023-12-28:
I am new to Modbus communication, and I am trying to learn it. I am using LPC1768 as the master and the Raspberry Pi as the slave for Modbus communication. Everything is working fine when I use the second version of the liblightmodbus library. When using the third version, even though no error is there, when I call the modbusMasterInit() function in my main loop, it's blocking the microcontroller from doing anything. Thus, I am not getting any output. Can you tell where I am going wrong?

I am adding a snippet of my code for reference.

`int main(void)
{   
	SystemInit();
	debugInitialize(DEBUG_BAUDRATE);
	UART_Init(C_UartOne_U8, MODBUS_BAUDRATE);
	
	ModbusErrorInfo err;
	err	= modbusMasterInit(
                    &master,          
                    dataCallback,                  		// Callback for handling incoming data
                    masterExceptionCallback,					// Exception callback (optional)
                    modbusDefaultAllocator,         	// Memory allocator used to allocate request
                    modbusMasterDefaultFunctions,   	// Set of supported functions
                    modbusMasterDefaultFunctionCount	// Number of supported functions
                    );
	assert(modbusIsOk(err)); 	
	
	osKernelInitialize();	
	
	tid_thrMODBUS_COMM_Q = osThreadNew(MODBUS_COMM_Q, NULL, &MODBUS_COMM_Q_attr);
	
	osKernelStart();

	return 0;
}`

Jacajack, 2024-01-03:
Hi and sorry for my late response. I have to say I'm not quite sure what could be the issue here. There's no reason for `modbusMasterInit()` to fail really. It only calls `modbusBufferInit()` and it never fails:

https://github.com/Jacajack/liblightmodbus/blob/dd3b242af1bd63e56a15dd8c97e03490ac259638/include/lightmodbus/base.impl.h#L51-L62

Perhaps your code is hanging at this assertion for some strange reason: `assert(modbusIsOk(err));`? Could you please see what happens when you remove it? Did you try printing anything right after the `modbusMasterInit()` call?

PoojaEmbedded, 2024-01-04:
Hi and thanks a lot for replying to my query.

As suggested, I tried the code after removing assert(modbusIsOk(err)), but
I'm still facing the same problem.
And yes, I tried printing after modbusMasterInit(), but I am getting
nothing at all.

I am sharing some screenshots for your reference. It would be a great help
if you could guide me through this.
For context, I am writing the code for the master of CNC controller. When I
press the reset button on it, I get the desired result (without_fn.png)
which happens when the function is not included.
And when the function is included, I get nothing after the reset button is
pressed (with_fn.png).

Thanks for your help
Pooja

Jacajack, 2024-01-04:
That's interesting. Unfortunately, I can't see the attachments - it seems that they don't pass through GitHub's email system.

I assume that `master` is a global variable in your code. Could you please see what happens if you make it a local variable in the `main` function?

If the program keeps crashing, we'll need to dive deeper and see what internal code actually causes the error. You can start by replacing the return value in this function by `MODBUS_NO_ERROR()` and getting rid of the `modbusBufferInit()` call.

https://github.com/Jacajack/liblightmodbus/blob/dd3b242af1bd63e56a15dd8c97e03490ac259638/include/lightmodbus/master.impl.h#L79-L94

As far as I'm concerned, assignments such as these should never fail unless they are referring to some protected memory region or something like that. Sadly, I'm not familiar with your platform, so I don't know any of its quirks or limitations.

PoojaEmbedded, 2024-01-05:
That's really silly of me that I didn't attach the photos properly. 
Here I am posting them.
Sorry for the inconvenience. 
1) With the function included:
[image]
2) When the function is commented out.
[image]

And yes, you're right, `master` is global variable in my code.
I'll try out all the suggested solutions and  let you know if I got the desired result or not.

Thanks a bunch.

Jacajack, 2024-01-05:
Glad to hear that worked for you! I wonder what was wrong with `master` being global. I assume that must be something platform-specific.

closed 2024-01-05 by Jacajack